RoboCupULaval/StrategyAI #48: API InfoManager

agingrasc, 2016-07-28:
(modification ultérieur à la demande possible)

* Set player_skill_target_goal, à séparer en 3

AlexandraMercier, 2016-09-15:
Refonte du InfoManager:
- Séparer l'état du jeu de InfoManager: Ce sera un singleton avec un update et des "get" read-only.
- La gestion des modules intelligents se fera par une classe singleton, qui chargera les modules dans __init__ avec un fichier config.
- La refonte va causer des bris dans l'API, réparer les bris

AlexandraMercier, 2016-09-17:
Je suis prête à commencer l'intégration du Game State manager et du Module manager dans l'API. Toutefois, il me manque des modifications en pull request qui enlèvent des dépendances au info manager.

closed 2016-11-10 by agingrasc