RoboCupULaval/StrategyAI #60: Ajouter des options lignes de commandes

agingrasc, 2016-08-18:
Le lancement de l'IA devrait permettre de passer des paramètres diverses:

* --test_bench : lance un test_bench pour la simulation/vrai vie
* --no_debug : coupe les fonctionnalités de debug

Le point est d'avoir le code en place (pas très lourd puisque dans la stdlib) et de pouvoir par la suite ajouter rapidement et facilement d'autres options qui seraient utile.

MaximeGLegault, 2017-01-07:
Options de ligne de commande ajoutées sauf celles mentionnées plus haut.

MaximeGLegault, 2017-04-26:
Fichier de configuration remplace les lignes de commandes. Juste à mettre comme option au scipt main.py un des fichiers cfg qui se trouve dans le dossier config -> python main.py config/sim_kalman_redirect.cfg

closed 2017-04-26 by MaximeGLegault