RoboCupULaval/StrategyAI #74: Considérer un facteur d'échelle pour le terrain

agingrasc, 2016-09-16:
Lors des tests d'intégrations il a été remarqué que le terrain a un facteur d'échelle (réduit, ~3X).
Ceci fait en sorte que les **deadzone** et autres variables utilisés pour éviter les collisions ne sont plus adéquate.

Il faut modifier l'architecture pour prendre en compte ce facteur.

- [ ] Modifier les grandeurs du terrains en fonction de l'échelle
- [ ] Modifier les constantes selon le facteur d'échelle
- [ ] Reçevoir le facteur à partir du UI-Debug
- [ ] Déterminer un facteur par défaut et pouvoir démarrer l'IA en mode _intégration_

agingrasc, 2016-10-13:
Une implémentation simpliste serait de reçevoir le paquet du UI et de mettre une valeur de ratio qui modifie les constantes clef (rayon d'un robot et rayon de la balle), les autres constantes devraient être définie en fonction de celles-ci.

MaximeGLegault, 2017-04-26:
Done as start option in config file

closed 2017-04-26 by MaximeGLegault