RoboCupULaval/StrategyAI #99: Gestion des équipes et de leur couleur

agingrasc, 2016-11-01:
Actuellement nous avons hardcode la valeur que nous sommes l'équipe bleu.
Cette considération devrait être gérée de façon plus générale

- [ ] Modifiable par le UI
- [ ] Modifiable par le referee (lorsqu'implémenté)

Il faudrait donc en faire la discussion d'architecture et minimiser les impacts.

MaximeGLegault, 2017-01-12:
Les couleurs des équipes ne sont plus hardcodées. On peut passer un paramêtre ligne de commande pour changer la couleur des équipes lors du démarrage. On ne demande plus "notre_équipe_jaune?" c'est maintenant rendu "notre couleur == couleur x de l'enum couleur". 
Pour modifier à partir de l'UI cela ne devrait pas être trop dur. Le framework est presque rendu à utiliser les packets de débogages, il ne lui reste qu'à répercuter le changement de couleur dans le reste du programme à voir...
 Sinon, de l'arbitre est-ce que c'est necéssaire? Va-t-on ce faire demander de changer de couleur au milieu d'un match ou juste au début lors de notre initialisation? Le second cas est déjà couvert.

sim0504, 2017-06-08:
«Va-t-on ce faire demander de changer de couleur au milieu d'un match ou juste au début lors de notre initialisation? »
Il est possible de se faire demander de changer de couleur au milieu d'un match, mais on peut empêcher ca en donnant la couleur préférée à l'autre équipe...

closed 2017-07-14 by pturgeon30