RoboCupULaval/StrategyAI #209: Fix share configuration between UI-Debug and IA

PhiBabin, 2017-09-27:


sim0504, 2018-02-28:
Is this fixed with field config?

PhiBabin, 2018-02-28:
That specific issue is suppose to find a better way to handle the sharing of the configuration files. But I found that it's pretty much impossible to do without having Ai depend on UI-debug or vice-versa. 
Having Ai depends on an handshack with UI-debug could work. However, the benefit of having a single config file does not outweigh the cost of that overhead.

PhiBabin, 2018-09-30:
Présentement, pour lancer l'ia il faut écrire ceci: `python3 main.py config/sim.cfg blue positive`
Il serait préférable si en plus de pouvoir ouvrir un fichier, on peut nommer le fichier dans le dossier `config/`. Ainsi, `python3 main.py blabla blue positive` devrait vérifier si `config/blabla.cfg` existe avant d'essayer d'ouvrir un fichier nommer `blabla` dans le répertoire courant.
Je propose que ui-debug peut à la fois loader un fichier du dossier `config/` ou du dossier `config/field/`.