RoboCupULaval/StrategyAI #212: Prevent access to tactics not in tacticBook

david2194, 2017-10-03:


PhiBabin, 2017-10-04:
On pourrait faire comme UI debug fait pour la reception de packet: toutes les tactics qui sont le dossier tactics/ sont automatiquement rajouté au tacticBook. Ainsi, cela évite que l'on laisse des tactics que nous utilisons pas.

david2194, 2017-10-04:
Le problème que je voyais c'était plutôt l'inverse. Nos strategy passent pas par le tacticBook pour importer leur tactics, elles vont direct faire des `from ../Tactic/GoalKeeper import GoalKeeper`. Ca fait des imports partout, et si un jour ot strategy peuvent utilement hériter d'autres strategies, on va se ramasser à importer mer et monde à chaque fois

PhiBabin, 2017-10-04:
Tu vois un genre service locator pour tactic, qui utiliserait le tacticBook pour évité d'avoir un shit load d'import?

david2194, 2017-10-06:
Quelque chose du genre, oui. J'ai pas encore trop élaboré les détails plus techniques, mais au moins ca laisserais la gestion de l'import/localisation des tactics au TacticBook, et les Strategy pourraient se focuser sur ce qu'elles sont réellement sensées faire

PhiBabin, 2018-02-26:
Je vais close, puisque cela ne semble pas être un problème urgent. Reopen quand qu'un sera intéressé

closed 2018-02-26 by PhiBabin