RoboCupULaval/StrategyAI #215: Refactor IACommand

PhiBabin, 2017-10-12:
IACommand currently is just a big dictionnary. Each time we add a new command, we need to add new field to it. So the contructor become a monster and it breaks all the time.
The solution is either to pass a list of command instead of one big object or to have a Builder contruct a IACommand.

MaximeGLegault, 2018-01-25:
Branche refonte_communication: J'ai changé l'AICommand pour le namedtuple de Simon, déplacer l'ancien dans le fichier Util/ai_command_shit en attendant de faire les changement partout ailleurs que idle et go to position path.

Une factorie pour IACommand serait bien.

PhiBabin, 2018-02-22:
Je vais changé les actions par un builder de ai_command.

PhiBabin, 2018-02-24:
See #248

closed 2018-02-24 by PhiBabin