RoboCupULaval/StrategyAI #245: Ai crash quand pall est a (0,0) et qu'on call gokick avec grsim

wonwon0, 2018-02-22:
Ai crash quand pall est a (0,0) et qu'on call gokick avec grsim

je sais pas si le bug est consistant, mais quand la balle est a (0,0), le ball service delete plein de ball.

wonwon0, 2018-02-22:
Simon, je t'ai assign sur celle là just in case que tu as une idée pk c'Est de même, notify me si t'as pas le temps de check ca, je vais le faire si non.

sim0504, 2018-02-22:
peux tu preciser une branche please

wonwon0, 2018-02-22:
refonte_communication

sim0504, 2018-02-22:
Peux tu me donner le log dans la console

sim0504, 2018-02-22:
Je dois dire que le multiball service est broken depuis le merge dans l'ai. Faudrai que je le fix.

PhiBabin, 2018-02-26:
Présentement, il y a rien qui crash lorsque l'on fait les actions mientionné plus haut. Par contre, le robot ne peut pas kicker la balle parce qu'il y a des robots invisibles au centre du terrain qui sont considéré collisionable.

barch2712, 2018-02-26:
Je me met la dessus

sim0504, 2018-03-01:
afaik this is fix.

closed 2018-03-01 by sim0504