RoboCupULaval/StrategyAI #254: Make strategy less unreadable

PhiBabin, 2018-02-27:
Currently the state machine of strategy is defined in the following manner:
```python
                self.add_tactic(index, PositionForPass(self.game_state, player, auto_position=True))
                self.add_tactic(index, GoKick(self.game_state, player, auto_update_target=True))

                self.add_condition(index, 0, 1, partial(self.is_closest, player))
                self.add_condition(index, 1, 0, partial(self.is_not_closest, player))
                self.add_condition(index, 1, 1, partial(self.has_kicked, player))
```
This is quite unreadable and can lead to simple error. I suggest to remove the utilisation of id (0,1,2, etc,) and instead use the nodes object. For example, it could be done like this:
```python
                node_pass = self.create_node(index, PositionForPass(self.game_state, player, auto_position=True))
                node_go_kick =self.create_node(index, GoKick(self.game_state, player, auto_update_target=True))

                node_pass.connect_to(node_go_kick, if=partial(self.is_closest, player))
                node_go_kick.connect_to(node_pass, if=partial(self.is_not_closest, player))
                node_go_kick.connect_to(node_go_kick, if=partial(self.has_kicked, player))
```

sim0504, 2018-02-27:
if keyword argument seems like a bad idea ;) But the idea seems nice!

PhiBabin, 2018-02-27:
I will kill for a 
```python
node_pass.connect_to(node_go_kick).if(partial(self.is_closest, player))
```

sim0504, 2018-02-27:
`player_is_closest  = partial(self.is_closest, player)`
`node_pass.connect_to(node_go_kick, when=player_is_closest)`
:)

sim0504, 2018-03-15:
#352

closed 2018-04-01 by PhiBabin