RoboCupULaval/StrategyAI #265: Add variable kick speed

sim0504, 2018-03-01:
We need to handle the kick force. In GrSim it goes from 0 (no kick) to 5 (ball explode). In real life the values might be different. We need a common interface which handle the conversion according to the game_type real or sim. One way is to set a fix range of level in the AI and handle that in the engine with the robot command senders.

Example: AI use values from 0 to 5 for kick force. value isnsent to engine in a engine command. Kick command is sent to the robot command sender (grsim or serial) and convert to the correct value.

PhiBabin, 2018-03-01:
Currently in serial, the kick force == number of ms the kick pin is set to high

sim0504, 2018-03-14:
#351

closed 2018-03-14 by sim0504