RoboCupULaval/StrategyAI #315: Clean and repair parts of the GoBehind Action

frstp64, 2018-03-10:
> ./ai/STA/Action/GoBehind.py:53:        # TODO find something better MGL 2017/05/22 TODO repair MGL 2018/01/25
> ./ai/STA/Action/GoBehind.py:77:        # TODO: Remove this part of the logic, since we have a pathfinder to do all of that...
> ./ai/STA/Action/GoBehind.py:108:        # TODO why?!? MGL 2017/05/22

The problem: The current GoBehind action class has several things that need to be fixed:
- [ ] If a second position is not specified, the robot should not try to go between the opponent robot and the opponent's goal's middle position.
- [ ] The path logic should be removed from this class.
- [ ] The orientation of the destination position may not be necessary at all.

sim0504, 2018-04-16:
Action are dead.

PhiBabin, 2018-06-03:
No use anymore

closed 2018-06-03 by PhiBabin