RoboCupULaval/StrategyAI #343: Le controle du robot est GRANDEMENT affecté par les performances dun ordi

wonwon0, 2018-03-13:
On soupconne que ca l'à avoir avec la reception/traitement des frames de vision.

sim0504, 2018-03-13:
Je propose de logger le temps d'execution de certaines fonctions dans le code.

- [ ] Calcul du path
- [ ] Update des KalmanFilter
- [ ] Prediciton des KalmanFilter
- [ ] Execution du VelocityRegulator
- [ ] Envoi des commandes aux robots

sim0504, 2018-03-13:
Aussi, logger les fps de la vision.

frstp64, 2018-03-14:
Méthode suggérée: utiliser les outils de python faits pour faire du profilage sur le cpu time au lieux du clock time.

#344 pour référence

PhiBabin, 2018-03-14:
On pourrait essayer de calculé c'est quoi la latence du système. Comme cela on aurait deux indices sur la performance de l'IA.

sim0504, 2018-03-14:
Le profiling est foncitonnel sur dev. On va pouvoir tester ca.

closed 2018-03-23 by sim0504