RoboCupULaval/StrategyAI #354: Add a ball position predictor

sim0504, 2018-03-15:


PhiBabin, 2018-03-15:
Like a something that estimate where the ball will be base on physic?

sim0504, 2018-03-15:
A simple prediction module that take a dt as input and output the ball position in dt seconds according to its velocity. It could also check who is closer to the ball, how long it would take to reach it considering it's Velocity etc. 
Another feature could be to get a interception point on the field to catch the ball according to the robot position.

wonwon0, 2018-03-15:
ha je sais comment faire!

on a juste a checker l'intersection de la trajectoire de la balle avec l'ellipse qui représente les position possibles qu'on robot puisse atteindre en un certain dt.

apres avoir trouvé la solution, on valide si la balle est conetnue dans l'ellipse.

Si aucune solution est trouvée, on incrémente le temps d'interception et on recommence.

wonwon0, 2018-04-24:
finalement jai fais de quoi de vraiment vraiment plus simple. Ca fonctionne pour pour le go_kick, maybe on va re-open l'issue si on a besoin d'un vrai ball_position_predictor, mais for now, pour le go_kick ca fonctionne 100%

closed 2018-04-24 by wonwon0