RoboCupULaval/StrategyAI #370: Add collision zone

PhiBabin, 2018-04-07:
A player should be able to enter the goal area of the opposite team. Currently they can enter it.

wonwon0, 2018-04-09:
And thus was the initial purpopse of the collision_body

sim0504, 2018-04-13:
C'était rendu impossible à supporter. Un changement était nécessaire pour le futur.

sim0504, 2018-04-13:
J'ai commencé ;a faire le typing des obstacles (#feature/add_obstacles_type)

sim0504, 2018-04-13:
Il reste juste à determiner le avoid_dir pour les goals afin que le robot essaye pas de faire le tour du mauvais coté et à ajouter les constantes de field dans la creation de l'obstacle

PhiBabin, 2018-04-14:
Ouin, il faudrait refaire les constantes de field des goals et recheck tous les endroits dans le code qui assume un cercle. Ce ne sera pas une mince affaire...

sim0504, 2018-04-14:
Il faut quand meme les scaler en fonction du terrain

wonwon0, 2018-04-14:
Ça sera pas si pire, la plus grosse job serait de dealer ak dautre geometries

wonwon0, 2018-04-14:
Mon approche serait de faires une approximation ak des cercles pour avoir de quoi de fonctionnel, pis apres on inplemente les carrés si on a le temps

wonwon0, 2018-04-24:
javais pas lu que t'avais commencé a travailler dessus ca, sorry jme suis un-assign

wonwon0, 2018-06-01:
#403 , moi et simon on a déterminer que c'était pas nécessaire, on a juste a faire en sorte qu'aucun go_kick est commandé quand la ball est dans their_goal et que juste le goal_keeper ce fait demander de go_kick quand la balle est dans our_goal

closed 2018-06-01 by wonwon0