RoboCupULaval/StrategyAI #376: Adapt Grsim to the rules of division B

PhiBabin, 2018-04-18:
Currently the configuration of Grsim is made for division A. There was commit on the GrSim repo to handle switching between divisions, but it was removed. There currently an issue on the official repo, but it was not fixed: https://github.com/RoboCup-SSL/grSim/issues/66

PhiBabin, 2018-04-18:
Pull request done, you can use it while we wait for a merge: https://github.com/RoboCup-SSL/grSim/pull/68

PhiBabin, 2018-04-19:
The official repo has merge the pull request.

closed 2018-04-19 by PhiBabin

wonwon0, 2018-04-19:
Such proud, many robocup, very code

sim0504, 2018-04-24:
@PhiBabin  ta prochaine issu a détruire: https://github.com/RoboCup-SSL/grSim/issues/39

PhiBabin, 2018-04-24:
Wtf, il y a genre 3 PR aucune de mergé tous au tours de l'overlap.