RoboCupULaval/StrategyAI #384: Robot's speed is badly estimated in the positive y

PhiBabin, 2018-04-27:
When robot go from a point in the negative y (0, -1000) to a point in the positive y (0, 1000), with a speed of 1m/s, when the robot reach the central horizontal line, the kallman estimate a speed of 0.6 m/s.
The bug has not been tested in real life.
This bug might had exist in the previous architecture.

wonwon0, 2018-05-09:
well, c'est pas juste dans les y positifs, jai tester en real life
j'ai fix ca sur la branche fix/motion_control_kalman en changeant les covariances en fixant le fps de l'engine qui changeait pas pour vrai malgrés l'argument quand on lancait l'ia (on manque plein de frame si on roule pas au meme fps que la vision so ca aide pas vraiment...)

PhiBabin, 2018-05-09:
Ouin il y a un bug où la commande line des fps était pas utilisé. Le framerate de l'engine n'influence pas le nombre de frame reçu. Le tracker handle les frames à mesure qu'il arrive et update le kallman. L'engine lui change les commandes de robot basé sur l'estimé du kallman.

wonwon0, 2018-05-31:
rekt...
resulu apres la PR de fix/motion_control_kalman

closed 2018-05-31 by wonwon0