RoboCupULaval/StrategyAI #398: Assuré que l'IA respecte les règles de Robocup SSL 2018

PhiBabin, 2018-05-27:
Il est important que notre IA respecte les règles du jeu pour évité des cartons jaunes/rouges. Il faudrait donc que soit en passe à travers les règles ou que l'on teste avec un autoref. 
Je n'ai pas encore réussie à faire marché un autoref avec notre IA.
Présentement voici la liste des règles que l'on désobéit:
 [ ] - Les robots rentrent dans la zone des buts adverses.

sim0504, 2018-06-06:
Lors d'un but, nos robots foncent vers l’intérieur du but. Je crois qu'il y a une tactic qui assign encore des positions à des endroits illégals

PhiBabin, 2018-06-07:
Ouin, je pense que c'est cause par ce que techniquement le goal_area n'incluent pas l'intérieur du but. Anyway en game le Play est stop si la balle rentre dans le but

PhiBabin, 2018-06-23:
On a eu zero carton jaune, fecque je pense que l'on respecte les règles...

closed 2018-06-23 by PhiBabin

PhiBabin, 2018-06-23:
C'était pourquoi déjà?

wonwon0, 2018-06-23:
((((((Optionnal))))))

sim0504, 2018-06-23:
too many defender in defence area mais c'était un edge case les robots etaient tous en tas devant le but.