RoboCupULaval/StrategyAI #406: La force du kick de go_kick devrait être fonction de la distance

PhiBabin, 2018-06-02:
Il serait utile que go_kick est un flag pour qu'il décide automatiquement la force du kick en fonction de la distance de la cible.

PhiBabin, 2018-06-03:
Genre

sim0504, 2018-06-08:
Ca serait bcp plus simple que l'enum du kick force contient une fonction static qui retourne un state selon la distance donné en paramètre

sim0504, 2018-06-08:
Sinon on enlève la kickforce et on passe la distance en paramètre directement

sim0504, 2018-06-08:
Par contre, ca va etre ultra variable la distance, tandis que la force est plus constante

PhiBabin, 2018-06-08:
Ouin lidee est que tu as besoin de la distance uniquement pour une passe et donc tu veux upper bound distance. Ce n'est pas grave  de tire pas assez  loin lors d'une  passe. Lors d'un  tir au but on veut tiré à puissance max.

sim0504, 2018-06-08:
on peut faire des états plus utile genre SHORT_PASS, LONG_PASS, BUMP, GOALKICK

PhiBabin, 2018-06-08:
Sa revien au même que d'avoir des LOW, MEDIUM et HIGH kick faire cela. C'est pour cela l'idée d'une valeur analogique, au lieu de discrète

PhiBabin, 2018-06-14:
Finalement, il a été décidé que tous les kicks au goal seront à max kick. Dépend de la distance de la passe le kick serait faible ou moyen

PhiBabin, 2018-06-14:
#474

closed 2018-06-14 by PhiBabin