RoboCupULaval/StrategyAI #412: Rajouté une option pour ignoré les balls sur côté de terrain

PhiBabin, 2018-06-04:
En compétition, le temps avec un terrain est précieux. C'est pourquoi il arrive souvent que deux équipes débug sur le même terrain, le problème avec deux équipes sur le même terrain s'est 2 balles sur le même terrain. Il est donc utile d'avoir un option dans l'IA pour _filter out_ la balle du côté positif du terrain ou du côté négatif.

sim0504, 2018-06-05:
Je vais travailler la dessus

PhiBabin, 2018-06-05:
Sur les outils de TIGERS, ils appellent cette option 'camera filtering'

sim0504, 2018-06-07:
Fonctionne sur multiball. Pas encore push. On peut désactiver les caméras qu'on veut pas

sim0504, 2018-06-07:
C'est push. J'ai pas vraiment fait un filtre spécifique pour les cotés du terrain, mais ca revient au meme. Si il y a une rogue ball qui se promène, elle va être considérer comme une nouvelle balle. On peut aussi desactiver les cameras avec leur id: `disabled_camera=[0,1,2,3]` dans le fichier de config

sim0504, 2018-06-07:
#426

closed 2018-06-07 by sim0504