RoboCupULaval/StrategyAI #432: Le contrôleur en vélocité arrête trop tôt à bas fps

PhiBabin, 2018-06-09:
Avec un fps dengine en dessous de 70 fps, dans la vraie vie, le contrôleur en vitesse va arrêter trop tôt ( 50cm ) avant sa destination.

sim0504, 2018-06-09:
Qu'est ce que tu veux dire? les commandes ne sont plus envoyé? le controlleur est reset? le controlleur ne s'exécute plus?

PhiBabin, 2018-06-09:
Le robot arrête à 50cm de la target. Le controler envoie un vitesse 0 quand il devrait envoyer une vitesse non-nulle.

wonwon0, 2018-06-10:
honetement fuck les bas fps

wonwon0, 2018-06-10:
maudite populace

sim0504, 2018-06-10:
Probablement que l'estimation de la vitesse est moins bonne. Faudrait tester avec un graph

PhiBabin, 2018-06-10:
Après des tests l'emergency break cause le bug: https://github.com/RoboCupULaval/StrategyAI/blob/dev/Engine/regulators/velocity_regulators.py#L37

sim0504, 2018-06-10:
En simulation je roule parfaitement à 20Hz. Probablement que le vrai robot freine bcp plus vite.

PhiBabin, 2018-06-10:
En simulation aussi il le fait. Il ne le fait pas tout le temps

sim0504, 2018-06-10:
J'ai fait beaucoup de test sur mon ordi et j'ai jamais vu ca à 20fps. c'est un nouveau problème donc?

sim0504, 2018-06-14:
seems to be solve afaik

closed 2018-06-14 by sim0504